davidramiro/Marlin-Ai3M #74: Mesh bed leveling

bartimau, 2020-03-24:
Hi!

My mesh bed leveling worked perfectly fine for a while but after i had to move the printer i don’t get it back to “normal”. 

I tryed the normal levelIng + mesh leveling already 10 times over again. The first layer is still not satisfying. 

Do u have any idea what could cause the prob? I read on some forums ppl changed back to 1.4.4 Bc of this.

Greetings

davidramiro, 2020-03-24:
Hi there,

that's hard to tell for me... I wouldn't suspect the software itself if it worked well in the beginning and doesn't after moving it. But of course, feel free to revert to 1.4.4 and if it fixes the issue, I know where to look.

Best regards,
David

kabukimandelux, 2020-03-28:
I do have an issue using 1.4.5 and mesh leveling. 

After the upgrade which I did twice now I followed the tutorial and did the leveling. When I print now, the nozzle is definately too high and on the display the Z value never changes on the initial layer.
I doubt mesh leveling is used although I tried to set up cura correctly:

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F{speed_travel} ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
M501
M420 S1
G1 F{speed_travel}
M117 Printing...
G5

`Send: G29 S0
Recv: State: On
Recv: Num X,Y: 5,5
Recv: Z offset: -0.05000
Recv: Measured points:
Recv:         0        1        2        3        4
Recv:  0 -0.04000 -0.04000 -0.06000 -0.08000 -0.08000
Recv:  1 -0.04000 +0.04000 +0.08000 +0.12000 +0.16000
Recv:  2 -0.08000 +0.04000 +0.06000 +0.16000 +0.24000
Recv:  3 +0.04000 +0.12000 +0.16000 +0.20000 +0.22000
Recv:  4 +0.16000 +0.18000 +0.12000 +0.08000 -0.02000
Recv: 
Recv: X:64.93 Y:189.61 Z:0.20 E:103.60 Count X:5195 Y:15169 Z:124
Recv: ok
`

I also changed G29 S4 to see if it has an effect, but I couldn't notice one.

Thanks for the great work, would love to know if this might be a firmware issue or my issue :)

kabukimandelux, 2020-03-28:
> I do have an issue using 1.4.6 and mesh leveling.
> 
> After the upgrade which I did twice now I followed the tutorial and did the leveling. When I print now, the nozzle is definately too high and on the display the Z value never changes on the initial layer.
> I doubt mesh leveling is used although I tried to set up cura correctly:
> 
> G21 ;metric values
> G90 ;absolute positioning
> M82 ;set extruder to absolute mode
> M107 ;start with the fan off
> G28 X0 Y0 ;move X/Y to min endstops
> G28 Z0 ;move Z to min endstops
> G1 Z15.0 F{speed_travel} ;move the platform down 15mm
> G92 E0 ;zero the extruded length
> G1 F200 E3 ;extrude 3mm of feed stock
> G92 E0 ;zero the extruded length again
> M501
> M420 S1
> G1 F{speed_travel}
> M117 Printing...
> G5
> 
> `Send: G29 S0 Recv: State: On Recv: Num X,Y: 5,5 Recv: Z offset: -0.05000 Recv: Measured points: Recv: 0 1 2 3 4 Recv: 0 -0.04000 -0.04000 -0.06000 -0.08000 -0.08000 Recv: 1 -0.04000 +0.04000 +0.08000 +0.12000 +0.16000 Recv: 2 -0.08000 +0.04000 +0.06000 +0.16000 +0.24000 Recv: 3 +0.04000 +0.12000 +0.16000 +0.20000 +0.22000 Recv: 4 +0.16000 +0.18000 +0.12000 +0.08000 -0.02000 Recv: Recv: X:64.93 Y:189.61 Z:0.20 E:103.60 Count X:5195 Y:15169 Z:124 Recv: ok `
> 
> I also changed G29 S4 to see if it has an effect, but I couldn't notice one.
> 
> Thanks for the great work, would love to know if this might be a firmware issue or my issue :)

Seems to work now. believe it was more an issue on my side.

Last question, i don't want to open another issue for the moment:

Since the upgrade from anycubic stock, my mix between 2208 drivers, and the stock for z axis, make a lot of noise compared to before, especially my y axis. Do you know what could be the reason ? Using stock driver .hex

bartimau, 2020-03-28:
Hi David,

its a bit weird. So after a few days of not printing, i startet my printer again (with out a new leveling) and it work. I didnt change anything just let it rest for a couple of days. I restarted it before after a fresh mesh leveling so i am wondering what went wrong. 

Greetings!

Feel free to close the ticket ;)

closed 2020-05-01 by davidramiro