davidramiro/Marlin-Ai3M #75: [FR] Repair the Thingiverse page.

Flo082002, 2020-03-25:
Hi,
I noticed that the Thingiverse side of this firmware is broken since Makerbot changed the design.
The text is no longer displayed correctly and all images are missing.
Maybe it can be repaired.

[image]

https://www.thingiverse.com/thing:3249319
Best regards 
Florian S

davidramiro, 2020-03-26:
Yup, seems like they broke Markdown formatting for texts this large. Nothing much I can do here. I tried moving it all into one section, eliminating the handy navigation bar, but it's still broken. 

Guess I'll have to create a page on my website and link that. Meh.

davidramiro, 2020-03-26:
Done. It's not beautiful, but it gets the job done: https://kore.cc/i3mega/doc/

Thanks for bringing this to my attention!

closed 2020-03-26 by davidramiro