davidramiro/Marlin-Ai3M #77: Marlin 1.1.9.1

stef-ladefense, 2020-04-12:
hi,

Marlin as release bugfix update, you have work :)

Best regards,

Stephane

philchuang, 2020-05-06:
Most of the changes are related to addition of korean language support, there are otherwise only a handful of files that have actual code changes. Looks like a lot of those have to do with BL touch, and it looks like the most important change is to G0/G1 behavior to reduce rounding error.

See diff here:
https://github.com/MarlinFirmware/Marlin/compare/1.1.9...1.1.9.1

stef-ladefense, 2020-05-06:
precisely, an update would benefit everyone, right?
it's a shame to have made the feat of brilliantly adapting the latest version of Marlin to this Anycubic screen, and leaving the last corrections behind either

davidramiro, 2020-05-06:
Then feel free to do a PR. I don't have much time to test the printer right now. I don't pull from origin without thorougly testing the features since there is quite a bit of custom code involved. Sorry, maybe I'll get the time next weekend.

Any issues with BLTouch that would require this update urgently?

stef-ladefense, 2020-05-06:
thank you David for your answer, and excuse me if my request seemed urgent.
no urgency not of course, I was just writing about the fact that there was a final fix for this almost dead version that is 1.9.1 and that it would be good to take advantage of it.

in any case thank you very much for your fabulous work.

I am pulling my rare hair right now to operate 2208 in uart mode, and even with RX on interrupt pins, it does not work completely, I will test with version 2 later

davidramiro, 2020-05-06:
>thank you David for your answer, and excuse me if my request seemed urgent.
no urgency not of course, I was just writing about the fact that there was a final fix for this almost dead version that is 1.9.1 and that it would be good to take advantage of it.

No worries, you didn't make it seem urgent, I was just curious if this was an important hotfix since yeah, 1.1.9 is pretty much EOL.

>I am pulling my rare hair right now to operate 2208 in uart mode, and even with RX on interrupt pins, it does not work completely, I will test with version 2 later

If you want my advice, ditch the Trigorilla... The time of 32bit boards has come anyways thanks to Marlin 2.

coding-pivo, 2020-10-27:
Hi,
I plan to play around a little bit with Marlin by implementing the fixes from Marlin 1.1.9.1. I have already forked your repo and would like to know how to go on in my forked repo to make it easy to open a pull request once I'm ready.
My plan is to generate a new branch in my forked repo to implement the changes. Would that work for a later pull request?
Thanks

davidramiro, 2020-10-27:
Hi,

>My plan is to generate a new branch in my forked repo to implement the changes. Would that work for a later pull request?

yes, sure, that would be a good way. If you have any specific questions, feel free to contact me.

coding-pivo, 2020-10-27:
So, I guess we can close this issue after merging my pull request? :)

davidramiro, 2020-10-27:
Indeed we can, Thanks a bunch!

closed 2020-10-27 by davidramiro