davidramiro/Marlin-Ai3M #78: [BUG] Hotend fan runs slower after 50°C

Dteif, 2020-04-12:
Hello, I replaced the stock hotend with an original e3d v6 hotend, with the e3d hotend fan. Everything runs perfectly apart from a problem that I'm not even sure is firmware related, but I'll try to post it here.

Everytime the hotend temperature reaches 50°C the fan is becoming clearly less noisy. I checked the voltages of the related pin, without anything connected, and it shows indeed 12.4V below 50°C and 7.4V above. This is explaining what happens with the fan running slower but not why.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Move axis by '....'
3. Start printing '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Photos of the print / output of the terminal**
If applicable, add photos or console outputs to help explain your problem.

**Printer (please complete the following information):**
 - Firmware version: [e.g. v19.02.08]
 - Printing method [e.g. SD, USB (OctoPrint, Cura, ...)]
 - Model [e.g. i3 Mega, Mega-S]

**M503 Output**
Please paste the complete output of the `M503` command here.

davidramiro, 2020-04-22:
Hi there, since you are running a custom fan, you would have to have a look at getting the PWM frequency right.

I wrote about that here: https://github.com/davidramiro/Marlin-Ai3M/wiki/Customization-&-Compiling#pwm-settings

Also, just to be sure, how are you measuring the voltage? A little piece of advice, never measure the voltage while the fan is not connected, because that will certainly fry the fan0 transistor on the board sooner or later. It's kind of a design flaw on the Trigorilla board. If you have to measure the voltage, always do it with a load.

davide-ricci, 2020-04-23:
Hi, thank you for replying! I saw some time ago that guide you wrote but I didn't think it could apply to this case, since my goal is to make the fan run at full speed (12 V), and therefore there shouldn't be any pwm involved, being a constant signal. Am I wrong? Also because below 50°C it works as expected at full speed.

I measured the voltage with a multimeter and if I'm correct I did it both with and without the fan connected, but thanks for the advice about the transistor. I didn't know it, and I'm curious about why that would happen.

Dteif, 2020-04-24:
Hi, thank you for replying! I saw some time ago that guide you wrote but I didn't think it could apply to this case, since my goal is to make the fan run at full speed (12 V), and therefore there shouldn't be any pwm involved, being a constant signal. Am I wrong? Also because below 50°C it works as expected at full speed.

I measured the voltage with a multimeter and if I'm correct I did it both with and without the fan connected, but thanks for the advice about the transistor. I didn't know it, and I'm curious about why that would happen.