davidramiro/Marlin-Ai3M #79: [FR] HEX for TMC2209

alexey-anufriev, 2020-04-27:
**Describe your suggestion.**
It would be nice to have precompiled hex binaries for TMC2209.
Another option is maybe a readme section explaining how to compile hex locally but also what to change to have it running with TMC2209.

**Reasoning**
It is not clear what to use with TMC2209. Stock version or TMC2208?

quattroerik, 2020-05-01:
I will have the same challenge, ordered some 2209 without checking prior what to do. Did you figure it out? Would be great if you could share it.
Thanks

davidramiro, 2020-05-01:
TMC2209 is fully backwards compatible. You can either use the stock or 2208 hex files. I cannot provide 2209 hex files since Marlin 1.x doesn't support that. Only since Marlin 2. (I have Marlin-Ai3M-2.0.x on my Git, although in early beta status)

closed 2020-05-01 by davidramiro