davidramiro/Marlin-Ai3M #80: M600 - Filament Change

cobhcdr66, 2020-04-27:
Hello,
I recently tried using the Filament Change feature by placing M600 at the layer in the .GCODE. After looking at your page (https://kore.cc/i3mega/doc/) with the detailed instructions you reference: " SD printing: Press CONTINUE (only once!) on the screen and wait for the hotend to heat up again" . I do not seem to have this "Continue" option on my screen and was wondering if you had any suggestions?

davidramiro, 2020-05-01:
As written in the docs, M600 via SD printing is beta and may not work reliably. If you are able to reproduce the issue, it'd be helpful if you could send over console logs when that happens (works best with TFT_DEBUG enabled in Configuration.h)

Baroudeur33, 2020-05-13:
I loooooove this firmware.
Just a tiny inconvenient, when i use change filament from SD print, all is correct, except at the end of printing, i dont have the message 'Printing done' and the button ok. I have to cancel the print, even it is completed to have back main screen options.