davidramiro/Marlin-Ai3M #82: BL going in wrong z direction

veevee2, 2020-05-09:
Hi,
I'm using the master source from 8/5/2020, and enabling the bi linear leveling for a piezo (FIX_MOUNTED_PROBE)

G28-Homing is ok : every axes goes to the right direction, even by distant control panel.
But G29-BL try to probe in the wrong z direction : it goes higher and higher.
I see no option related to the z direction of homing : what could be about related please ?

Thanks,
Olivier

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Move axis by '....'
3. Start printing '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Photos of the print / output of the terminal**
If applicable, add photos or console outputs to help explain your problem.

**Printer (please complete the following information):**
 - Firmware version: [e.g. v19.02.08]
 - Printing method [e.g. SD, USB (OctoPrint, Cura, ...)]
 - Model [e.g. i3 Mega, Mega-S]

**M503 Output**
SENT: M503
READ: echo:  G21    ; (mm)
G21    ; (mm)
READ: 
READ: echo:Filament settings: Disabled
Filament settings: Disabled
READ: echo:  M200 D1.75
M200 D1.75
READ: echo:  M200 D0
M200 D0
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X80.00 Y80.00 Z400.00 E92.60
M92 X80.00 Y80.00 Z400.00 E92.60
READ: echo:Maximum feedrates (units/s):
Maximum feedrates (units/s):
READ: echo:  M203 X500.00 Y500.00 Z6.00 E60.00
M203 X500.00 Y500.00 Z6.00 E60.00
READ: echo:Maximum Acceleration (units/s2):
Maximum Acceleration (units/s2):
READ: echo:  M201 X3000 Y2000 Z60 E10000
M201 X3000 Y2000 Z60 E10000
READ: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
READ: echo:  M204 P1500.00 R3000.00 T3000.00
M204 P1500.00 R3000.00 T3000.00
READ: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
READ: echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
READ: echo:Home offset:
Home offset:
READ: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
READ: echo:Auto Bed Leveling:
Auto Bed Leveling:
READ: echo:  M420 S0 Z0.00
M420 S0 Z0.00
READ: echo:Endstop adjustment:
Endstop adjustment:
READ: echo:  M666 Z0.00
M666 Z0.00
READ: echo:PID settings:
PID settings:
READ: echo:  M301 P15.94 I1.17 D54.19
M301 P15.94 I1.17 D54.19
READ: echo:  M304 P251.78 I49.57 D319.73
M304 P251.78 I49.57 D319.73
READ: echo:Z-Probe Offset (mm):
Z-Probe Offset (mm):
READ: echo:  M851 Z0.00
M851 Z0.00
READ: echo:Linear Advance:
Linear Advance:
READ: echo:  M900 K0.00
M900 K0.00
READ: echo:Filament load/unload lengths:
Filament load/unload lengths:
READ: echo:  M603 L538.00 U555.00
M603 L538.00 U555.00

veevee2, 2020-05-09:
I finally found the reason, but not a so clean solution.
Whatever is add "#define Z_MIN_PROBE_PIN 11" in configuration.h (11 is a pin slot available and connectable on the trigorilla 1, it's 5 on a trigorilla 1.1), it's overided by the include pins_TRIGORILLA_14.h which set "#define Z_MIN_PROBE_PIN 2"

I modified pins_TRIGORILLA_14.h but don't like this way, any cleaner solution ?

LordDarkneo, 2020-05-24:
Hi,
I am not using this mod for BL Touch (but the original Marlin). I have just made a diff with mine (using BL touch also). The main difference is this line commented:
//#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe.
Could you give a try?

veevee2, 2020-05-25:
Hi,
Finally tue problem is solved : too much sensibility of the piezo cause this continus z move until the sensibility is better adjusted.
Since that : everything World great (an accuracy is impressive)

closed 2020-05-25 by veevee2