davidramiro/Marlin-Ai3M #84: Bed heat shutoff

SeventhDread, 2020-06-07:
Hi, this happens with the Anycubic Firmware too, but I am currently using Marlin and like it, so I would love if you can help me also this is not really a bug but I thought it suited bug more than a feature request.

I am trying to print with ABS but when I try to heat the bed to 110C when it reaches like 102C I get this in terminal through octoprint: 
[image]
I believe this is to do with a timer because it takes too long to heat up, and I am wondering sence you made the Anycubic Marlin firmware if you know how to I edit something to extend this time?

Anycubic i3 Mega-S
 - 1.4.6 Marlin 1.9.9
 - Printing Method: Octoprint
 

**M503 Output**
ecv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E92.60
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E60.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y2000 Z60 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P4000.00 R3000.00 T4000.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:  G29 S3 X1 Y1 Z-0.22000
Recv: echo:  G29 S3 X2 Y1 Z-0.22000
Recv: echo:  G29 S3 X3 Y1 Z-0.16000
Recv: echo:  G29 S3 X4 Y1 Z-0.20000
Recv: echo:  G29 S3 X5 Y1 Z-0.14000
Recv: echo:  G29 S3 X1 Y2 Z-0.22000
Recv: echo:  G29 S3 X2 Y2 Z-0.22000
Recv: echo:  G29 S3 X3 Y2 Z-0.22000
Recv: echo:  G29 S3 X4 Y2 Z-0.20000
Recv: echo:  G29 S3 X5 Y2 Z-0.16000
Recv: echo:  G29 S3 X1 Y3 Z-0.16000
Recv: echo:  G29 S3 X2 Y3 Z-0.22000
Recv: echo:  G29 S3 X3 Y3 Z-0.18000
Recv: echo:  G29 S3 X4 Y3 Z-0.14000
Recv: echo:  G29 S3 X5 Y3 Z-0.14000
Recv: echo:  G29 S3 X1 Y4 Z-0.10000
Recv: echo:  G29 S3 X2 Y4 Z-0.12000
Recv: echo:  G29 S3 X3 Y4 Z-0.12000
Recv: echo:  G29 S3 X4 Y4 Z-0.10000
Recv: echo:  G29 S3 X5 Y4 Z-0.16000
Recv: echo:  G29 S3 X1 Y5 Z-0.02000
Recv: echo:  G29 S3 X2 Y5 Z-0.06000
Recv: echo:  G29 S3 X3 Y5 Z-0.06000
Recv: echo:  G29 S3 X4 Y5 Z-0.08000
Recv: echo:  G29 S3 X5 Y5 Z-0.10000
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P15.94 I1.17 D54.19
Recv: echo:  M304 P251.78 I49.57 D319.73
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00
Recv: ok

davidramiro, 2020-06-07:
Yes, you can change the timers:

https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/Configuration_adv.h#L100

Raise `THERMAL_PROTECTION_BED_PERIOD` and `WATCH_BED_TEMP_PERIOD`, lower `THERMAL_PROTECTION_BED_HYSTERESIS` and `WATCH_BED_TEMP_INCREASE`.

Since this is not really a bug, I am closing this issue, but feel free to report back with any further questions.

closed 2020-06-07 by davidramiro

SeventhDread, 2020-06-07:
Hi, thanks but how much do I increase/decrease them by?

SeventhDread, 2020-06-07:
Also when trying to upload it I got this error message: (I put it in a notepad file as its long)
[errorreport.txt](https://github.com/davidramiro/Marlin-Ai3M/files/4741999/errorreport.txt)

davidramiro, 2020-06-07:
Hard to say. Ideally, if you're printing ABS you should have the printer in a well insulated enclosure as well as having the underside of the bed insulated too (I am using ceramic fiber for that). 

```
#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 60    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 2 // Degrees Celsius

  /**
   * As described above, except for the bed (M140/M190/M303).
   */
  #define WATCH_BED_TEMP_PERIOD 90                // Seconds
  #define WATCH_BED_TEMP_INCREASE 1               // Degrees Celsius
#endif
``` 

Something like this should work.

davidramiro, 2020-06-07:
> Also when trying to upload it I got this error message: (I put it in a notepad file as its long)
> [errorreport.txt](https://github.com/davidramiro/Marlin-Ai3M/files/4741999/errorreport.txt)

Did you select the Mega2560 as the board/CPU in Arduino IDE?

Please see the following link for compilation instructions: https://github.com/davidramiro/Marlin-AI3M/wiki/Customization-&-Compiling

SeventhDread, 2020-06-07:
I had it on Ardunio Uno, and how do I save it as a hex file as when I click export complied binary it doesn't give me anywhere to save it just verifies it basically. If this does save it, where does it save it?

SeventhDread, 2020-06-07:
And I just have mine in a unused room (usually cold) and just leave it in there I normally print PLA so its good never printed ABS so I have not done anything to the printer no heat insulators or anything

SeventhDread, 2020-06-07:
Hi I used this link to export as hex https://www.instructables.com/id/HOW-TO-GET-HEX-FILE-FROM-ARDUINO-/ will this work?

SeventhDread, 2020-06-07:
Well the hex file the home completely was off it was breaking and all sorts, how do I get the stock drivers??

davidramiro, 2020-06-07:
Did you read the page I linked before? https://github.com/davidramiro/Marlin-AI3M/wiki/Customization-&-Compiling

>the home completely was off it was breaking and all sorts

You have to set the correct motor direction as explained in the Wiki.

>how do I save it as a hex file as when I click export complied binary

This is also explained in that page.

>how do I get the stock drivers??

This can also be found on [the wiki](https://github.com/davidramiro/Marlin-AI3M/wiki/Frequently-Asked-Questions#how-to-go-back-to-the-manufacturer-firmware).

Please make sure to read everything thoroughly, especially since you are changing temperature safety settings.

SeventhDread, 2020-06-07:
Ahh, I didn't read the part were you have to comment things, thank you, Will let you know if there is any more problems.