davidramiro/Marlin-Ai3M #90: Y axis crashing beyond boundaries after print cancellation

tarmogr, 2020-08-02:
The scenario is:

1. Print something and wait till it completes (im not sure if this step is necessary)
2. move the hotplate manually in some random position
3. Start a print job
4. Before the heaters are hot cancel the job by pressing the "Stop" button on the screen

Result: The print head will raise a little and then the Y axis will quickly move forward beyond limits with belts rattling.

I think that since the print was cancelled before it did the homing routine, it did not know the actual position of the hot plate. I think there should be a check somewhere so that if homing is not done, no positive movement would be allowed.

davidramiro, 2020-08-02:
Interesting, I did infact [include a check](https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/AnycubicTFT.cpp#L374) for that:

```
void AnycubicTFTClass::ParkAfterStop(){
  // only park the nozzle if homing was done before
  if (!axis_unhomed_error()) {
    ...
```

I'll have to do a bit more digging what actually counts as "not homed" in this context. Thanks for reporting this!

tarmogr, 2020-08-02:
Thanks for such a fast reply, let me know if you can reproduce it, if not I will try to recreate what I did to make that happen in a safe manner again.