davidramiro/Marlin-Ai3M #93: [BUG]

calufax, 2020-09-12:
Hola David, por tu nombre puedo deducir que hablas español, sino avisame y te traduzco al ingles. Tengo una anycubic i3 Mega S, hace una semana, de las que traen la pantalla de la MEgaX.
El problema es el siguiente: cuando cargo mas de 3 archivos en la tarjeta SD, solo muestra los 3 primeros. Actualice a tu version 1.1.9. No me paso antes de eso, pero no puedo asegurar que solamente debido a flashear el firmware. De hecho cuando no me reconocio el cuarto archivo recien cargado, borre los tres primeros y lo reconocio.
desde ya muchisimas gracias, aprecio el trabajo de tu equipo!! saludos

davidramiro, 2020-09-15:
La nueva pantalla MegaX no es totalmente compatible con este firmware. No tengo ese modelo, así no puedo hacer nada al respecto en este momento.

closed 2020-09-15 by davidramiro

kaneyan1022, 2020-12-13:
@calufax 
>Tengo una anycubic i3 Mega S, hace una semana, de las que traen la pantalla de la MegaX.

Me too!
Has this issue been resolved?
How did you solve this problem?
I am asking ANYCUBIC how to change from the MegaX screen to the MegaS screen. 
No answer has come at this time.