davidramiro/Marlin-Ai3M #96: [BUG] filament runout - No reaction

thomas41587, 2020-10-01:
**Describe the bug**
When the filament is empty, the TFT display reports that the filament sensor detected no filament. But the print just continues like before. I do not get any notification or similar in Octoprint. 
I am not sure if thisfeature is only limited to USB-printing or maybe I missed a configuration thing within Octoprint itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Start a print via Octoprint
3. Cut the filament so there is no filament within the sensor

**Expected behavior**
The print pauses

**Printer (please complete the following information):**
 - Firmware version: latest precompiled from this repository
 - Printing method OctoPrint
 - Model Mega

**M503 Output**
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E397.78
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E60.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y2000 Z60 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1500.00 R3000.00 T3000.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P15.94 I1.17 D54.19
Recv: echo:  M304 P251.78 I49.57 D319.73
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00
Recv: ok

davidramiro, 2020-10-01:
Please make sure to use a search engine or read through previously closed issues before reporting a bug here.

This is not a bug, the filament sensor simply cannot stop the print if you use a USB host. You could connect it to the Raspberry'S GPIO pins and use a plugin for OctoPrint, then the sensor would be able to interrupt the print.

closed 2020-10-01 by davidramiro

EnderSyth, 2021-08-29:
David I think this is incorrect, after doing some research I found this discussion that states you need to uncomment the lines for ACTION_ON_PAUSE and ACTION_ON_RESUME in configuration_adv.h to send the M118 A action:pause for this to work.

https://community.octoprint.org/t/octoprint-windows-pause-octoprint-when-printer-sends-m600-command/13067/3

I have not tested this however.