davidramiro/Marlin-Ai3M #101: Extruder stops in the first layer and continues in the second layer

Perrunge, 2020-11-09:
Hi there,

I have upgraded the firmware to 1.1.9. I completed the guide and everything seems to work... until I start printing. 

For some reason the extruder turns off in the middle of printing the skirt. The printer head continues to complete the coordinates of the first layer. Then in the second layer, the extruder starts again... When the extruder stops the motor is locked. Can't even manually feed filament.

Can it be the Z-endstops being activated in the first layer causing the extruder to stop?

Can anybody help me here?

davidramiro, 2020-11-10:
Hi,

I can't think of anything within the firmware that would cause this behaviour and I've never had a report like this, especially not on one particular layer. But if it happens right after you upgraded, and if the problem goes away after you reflash the manufacturer firmware, we have to look into this.

To start off, which slicer do you use?

Perrunge, 2020-11-10:
Hi David,

I use Cura. Same as I have used the past year.

I will try to reflash the original firmware.

davidramiro, 2020-11-10:
You can also try using this Cura profile: 
[cura-profile.zip](https://github.com/davidramiro/Marlin-Ai3M/files/5518925/cura-profile.zip)

I created it specifically for this firmware with some tweaked settings. In theory, every profile should work fine, but this way you could sort out a slicer issue.

Perrunge, 2020-11-10:
Thanks! I will try it ASAP

davidramiro, 2020-12-28:
Any progress on this? I'll be closing this for now, if there's still an issue feel free to report back and I'll reopen this.

closed 2020-12-28 by davidramiro