davidramiro/Marlin-Ai3M #102: [BUG] Info screen on LCD shows wrong firmware version

diegomgarcia, 2020-11-11:
After flashing the firmware and do all steps M502 and M500 command, restarted, et all. When you go to the info menu it says firmware version 1.2.0 the stock version.

Steps to reproduce the behavior:
1. Flash this firmware
2. Restart the printer
3. Go to tools, more, info
4. See the wrong firmware version

**Expected behavior**
Should show this version.

**Printer:**
 - Model Mega-S

**M503 Output**
Send: M503
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E384.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E30.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y2000 Z60 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1500.00 R1500.00 T3000.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P15.94 I1.17 D54.19
Recv: echo:  M304 P251.78 I49.57 D319.73
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00
Recv: ok

davidramiro, 2020-11-11:
https://github.com/davidramiro/Marlin-Ai3M/wiki/Frequently-Asked-Questions#the-display-shows-a-wrong-firmware-version

closed 2020-11-11 by davidramiro

diegomgarcia, 2020-11-11:
thanks didnt saw that FAQ available, it already answer another questions.

davidramiro, 2020-11-11:
Glad it could be of use!