davidramiro/Marlin-Ai3M #105: BUG Firmware v1.4.6 stable (Marlin 1.1.9) Mesh Bed Leveling I3 Mega  Z Problem 1.4.4 works fine?

raindancer2204, 2020-12-28:
**English:**
Hi there,
I really have to praise the development work here.

I have an I3 Mega.

However, I noticed the following with Mesh Bed Leveling.

Recently I have done the Mesh Bed Leveling according to the instructions several times.

I recently had a repair and had to remove my heating bed.

1. Back heated bed
2. Moved to the home position
3. Manually level the bed to Homig Position.
4. Mesh levling performed saved and okay. Machine off and on (reboot).

Every time I printed was the Z-axis at a different height? The values ​​for the more active mesh levling were in the G-CODE. Both via Octoprint and on an SD card.

I also did the rest using the control code and tried leveling several times.
There seems to be something wrong with the firmware?

There is also a post about this from my colleagues in the 3D printing forum.

https://drucktipps3d.de/forum/topic/anycubic-i3-mega-mesh-bed-leveling-problem/
I have now also gone to 1.4.4 and lo and behold, it works perfectly.

Can someone adjust that, please? If you have any questions, I am always available.

Thank you.

**Deutsch:**
Hallo,
ich muss die Entwicklungsarbeit hier wirklich loben.

Ich habe einen I3 Mega.

Jedoch ist mir folgendes aufgefallen beim Mesh Bed Leveling.

Ich habe in letzer Zeit öfters mal das Mesh Bed Leveling laut Anleitung gemacht. 

Ich hatte letzens eine Reperatur und musste mein Heizbett entfernen. 

1. Heizbett zurück
2. In die Home postion gefahren
3. Manuell das Bett auf Homig Postion gelevelt.
4. Mesh Levling durchgeführt gespeichtert und okay. Maschine aus und angeschaltet (Reboot).

Jedes mal wenn ich druckte war die Z-Achse auf einer anderen Höhe? Die werte für das aktiveren des Mesh Levlings waren im G-CODE. Sowohl über Octoprint als auf SD Karte.

Habe auch den Restet durchgeführt über den Steuercode und mehrmal das Leveln probiert.
Es scheint irgendwas mit der Firmware nicht zu passen? 

Es gibt hierzu auch einen Post von den Kollegen aus dem 3D Druck Forum.

https://drucktipps3d.de/forum/topic/anycubic-i3-mega-mesh-bed-leveling-problem/
Ich bin nun auch auf die 1.4.4 gegangen und siehe da es funktioniert perfekt.

Kann das bitte mal jemand nachstellen? Wenn fragen sind stehe ich jederzeit zur Verfügung.

Danke euch.

davidramiro, 2020-12-28:
Thorougly comparing those versions, there are i 2 small changes regarding G29/MMBL. Since I cannot reproduce this issue myself, as I am using v1.4.6 for months without any issues with MMBL, would you be able to test a custom build with a small change I'd compile for you? If so, I'd need to know which version of the .hex file you need. (TMC2208, reversed or stock)

raindancer2204, 2020-12-28:
**English:**
Hi David,

I would test it and also make an entry in drucktipps3d.de if that is okay with you? So that we have a "pair of testers" more who also reported it?

I myself need the version for the TMC2208 as a "reversed" version, and I can also test it.

I assume that the error can be reproduced by "multiple bedlevelig" (executing the level action save and switch off).

Otherwise I did this once every 6 to 8 weeks. Since the heating bed repairs try more often.

I try it. Should I activate the other testers? Would that be okay with you?

Greetings Raini

**Deutsch:**

Hi David,

ich würde es testen und auch einen Eintrag im drucktipps3d.de machen wenn das für dich okay ist? Damit wir ein "Paar Tester" mehr haben die es auch reportet haben?

Ich selbst benötige die Version für die TMC2208 als "Reversed" Version, und ich kann es auch antesten.

Ich gehe davon aus das der Fehler durch "mehrfaches Bedlevelig" (durchführen der Levelaktion Speichern Auschalten) reproduziert werden kann.

Sonst habe ich das alle 6 bis 8 Wochen eimal gemacht. Seit dem Heizbett Reperatur versuch öfters.

Ich probiere es. Soll ich die anderen Tester akivieren? Wäre dir das recht?

Grüsse Raini

davidramiro, 2020-12-28:
I've compiled test binaries for all three hex versions, so you can freely share them among other testers. I'll also keep an eye on the Drucktipps3D thread, I'm also a member on that forum.

[Marlin-Ai3M-MMBLtest.zip](https://github.com/davidramiro/Marlin-Ai3M/files/5748235/Marlin-Ai3M-MMBLtest.zip)

raindancer2204, 2020-12-28:
Thanks for your quick help. I will try the new test firmware tomorrow. Info the Thread in the forum:
https://drucktipps3d.de/forum/topic/anycubic-i3-mega-mesh-bed-leveling-problem/
was updatet. So lets see what the other are testing 👍 . I will check it tomorrow.

davidramiro, 2021-02-14:
Any further feedback? If that does indeed fix the issue, I would like to merge this into the firmware and release a new version.