davidramiro/Marlin-Ai3M #106: [BUG]SD Card Menu will not show

3dRikal, 2021-01-05:
**Describe the bug**
The special menu for bed leveling will not appear when the SD Card is pushed in.

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Insert freshly formatted 4GB SD Card (formatted in FAT32 with SDCARD.ORG tool)
3. No menu appears after hitting refresh

**Expected behavior**
Menu should appear

**Notes**
I have tried pulling the SDCARD a little, a lot, reinserting. Nothing works. The SD Card reader does work as I can print from it when putting files on the card. There is noting at the moment on the SD Card (no funky characters)

**Printer (please complete the following information):**
 - Firmware version: 1.1.9
 - Printing method Octoprint (primary) SD, Cura
 - Model: Mega-S

**M503 Output**
`Send: M503
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E384.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E60.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000 Y2000 Z60 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1500.00 R3000.00 T3000.00
Recv: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P15.94 I1.17 D54.19
Recv: echo:  M304 P251.78 I49.57 D319.73
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L538.00 U555.00
Recv: ok
`

davidramiro, 2021-01-05:
I have had reports that Anycubic is now using a new screen on older models, on which the SD card menu will not work.

Can you confirm if your screen looks something like this:

[image]

3dRikal, 2021-01-05:
It does indeed. If you need me to beta test a fix, i will volunteer.

[86528432-f526d480-bea7-11ea-80a1-f808cda3c9f6]<https://user-images.githubusercontent.com/30576493/103652547-8cf2a180-4f63-11eb-95d6-be5e4d5c946e.jpg>

davidramiro, 2021-01-05:
Sorry, can't really look into it at the moment due to lack of time. Although I can recommend this firmware which supports this screen: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/

3dRikal, 2021-01-05:
Thanks for the recommendation. It worked. The DGUS hex worked and I now can level my bed. Sent from Mail for Windows 10 From: David RamiroSent: January 5, 2021 9:25 AMTo: davidramiro/Marlin-Ai3MCc: 3dRikal; AuthorSubject: Re: [davidramiro/Marlin-Ai3M] [BUG]SD Card Menu will not show (#106) Sorry, can't really look into it at the moment due to lack of time. Although I can recommend this firmware which supports this screen: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/

7anner, 2021-01-15:
Just wanted to note that I had issues that extended beyond just the SD menu with this firmware.  Notably, both times that I tried to print, the x axis would immediately attempt to overextend, and then the printing would halt.

The alternative firmware you recommended does work pretty good.

Thanks.

davidramiro, 2021-01-15:
> Just wanted to note that I had issues that extended beyond just the SD menu with this firmware. Notably, both times that I tried to print, the x axis would immediately attempt to overextend, and then the printing would halt.

Which printer model do you use? I am curious whether they changed something else than the screen on i3 Mega/Mega-S model. I do know that the Mega X/Pro has other drivers, so this behaviour would be expected.

7anner, 2021-01-16:
It's a Mega S.  Anything you want me to check?