davidramiro/Marlin-Ai3M #108: [BUG]Pin numbering for z2_endstop not consistent in source

Gleis1234, 2021-01-11:
**Describe the bug**
I was adding BLTouch to my I3 Mega (Upgraded to S). After making the wiring and adding two MOSFET-drivers for Hotend and Hotbed, I got a problem with the z_endstops. Everytime the left z_endstop was activated, the printer makes a shutdown. The reason was the D42 plug, which was inserted between 5V and GND. After plugging it correctly only the left z-servo stops in a home operation, the right one continues to turn, when hitting the endstops. Both are leveled at the same height. Is it possible that both z-servos are interchanged? I read this idea in another forum.
Therefore I tried to follow all wires from the hub to the motherboard. I noticed, that the left z_endstop is connected with a 2-wire cable to D42 and the y_endstop with a 3-wire cable to D43. In the pins_TRIGORILLA_14.h these two connections are defined vice versa,  D42 is assigned to Y_MIN_PIN and D43 to X_MAX_PIN . I installed the precompiled .hex file and the assignment seems to be correct, M119 shows the correct triggers. Is this only an issue of the source and has to be changed before compiling? Has the logic for the y_min_endstop be inverted, because the switch is connected to 5V?
I am totally confused.

**To Reproduce**
Steps to reproduce the behavior:
1. Home z_axis

**Expected behavior**
Both z-servos should stop

Explanation of correct pin assignment.

**Printer (please complete the following information):**
 - Firmware version: 1.1.9 (1.4.6)
  - Model  [Mega-S]

**M503 Output**
Please paste the complete output of the `M503` command here.

davidramiro, 2021-01-11:
I have no clue about BLTouch, and I can't really help you out there as I do not know what exactly you changed that made your second Z endstop fail. 
What I do have a clue about, is that the silkscreen on the motherboard, e.g. the numbering of pins, does not correspond exactly to the RAMPs pins. This firmware is configured fine for what it advertises, even though for example there is a discrepancy between E1 and Z for the drivers. Also internally, the 2nd Z endstop is actually mapped to X_MAX.

Also, the builds are directly compiled from source.

As mentioned in the documentation, head over to this repository for a BLTouch firmware based on this one: https://github.com/MNieddu91/Marlin-AI3M-BLTouch. There are wiring instructions on the documentation that will definitely not cause the endstops to fail.

closed 2021-01-11 by davidramiro

Gleis1234, 2021-01-12:
Thank you for your quick response. Good to now that you can't trust the silkscreen but the source.