davidramiro/Marlin-Ai3M #110: Hotbed heating stops after about 10 minutes printing

Gleis1234, 2021-01-24:
**Describe the bug**
Running a print job, no matter if started from SD or over USB, after about 10 minutes the hotbed temperature gets lower, and after falling below a certain limit the firmware stops with a thermal runaway.

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
2. Start printing '....'
3. After some minutes, normally within first layer, hotbed temperature heating stops and firmware is halted with thermal runaway. Hotend temperature keeps stable.

**Expected behavior**
Heatbed temperature keeps stable. 

**Photos of the print / output of the terminal**

**Printer (please complete the following information):**
 - Firmware version:Marlin 1.1.9 (release 1.4.6)
 - Printing method : No matter, if SD or USB with Repetier host
 - Model : Mega-S

**M503 Output**
Send:13:57:19.944: N13 M503
Recv:13:57:19.949: echo: G21 ; (mm)
Recv:13:57:19.949: echo:Filament settings: Disabled
Recv:13:57:19.951: echo: M200 D1.75
Recv:13:57:19.952: echo: M200 D0
Recv:13:57:19.952: echo:Steps per unit:
Recv:13:57:19.954: echo: M92 X80.00 Y80.00 Z400.00 E389.30
Recv:13:57:19.955: echo:Maximum feedrates (units/s):
Recv:13:57:19.957: echo: M203 X500.00 Y120.00 Z6.00 E60.00
Recv:13:57:19.960: echo:Maximum Acceleration (units/s2):
Recv:13:57:19.960: echo: M201 X2000 Y1200 Z60 E10000
Recv:13:57:19.963: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv:13:57:19.965: echo: M204 P1200.00 R3000.00 T1500.00
Recv:13:57:19.971: echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv:13:57:19.973: echo: M205 Q50000 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv:13:57:19.974: echo:Home offset:
Recv:13:57:19.975: echo: M206 X0.00 Y0.00 Z0.00
Recv:13:57:19.975: echo:Auto Bed Leveling:
Recv:13:57:19.975: echo: M420 S0 Z0.00
Recv:13:57:19.978: echo: G29 W I0 J0 Z-0.16750
Recv:13:57:19.978: echo: G29 W I1 J0 Z-0.16000
Recv:13:57:19.980: echo: G29 W I2 J0 Z-0.16250
Recv:13:57:19.981: echo: G29 W I3 J0 Z-0.17000
Recv:13:57:19.983: echo: G29 W I4 J0 Z-0.18250
Recv:13:57:19.983: echo: G29 W I0 J1 Z-0.13000
Recv:13:57:19.986: echo: G29 W I1 J1 Z-0.12250
Recv:13:57:19.987: echo: G29 W I2 J1 Z-0.10750
Recv:13:57:19.987: echo: G29 W I3 J1 Z-0.11000
Recv:13:57:19.989: echo: G29 W I4 J1 Z-0.13250
Recv:13:57:19.989: echo: G29 W I0 J2 Z-0.12750
Recv:13:57:19.992: echo: G29 W I1 J2 Z-0.11750
Recv:13:57:19.992: echo: G29 W I2 J2 Z-0.11250
Recv:13:57:19.994: echo: G29 W I3 J2 Z-0.10500
Recv:13:57:19.995: echo: G29 W I4 J2 Z-0.12750
Recv:13:57:19.997: echo: G29 W I0 J3 Z-0.15750
Recv:13:57:19.997: echo: G29 W I1 J3 Z-0.13250
Recv:13:57:20.000: echo: G29 W I2 J3 Z-0.12000
Recv:13:57:20.000: echo: G29 W I3 J3 Z-0.13000
Recv:13:57:20.000: echo: G29 W I4 J3 Z-0.14500
Recv:13:57:20.003: echo: G29 W I0 J4 Z-0.21500
Recv:13:57:20.003: echo: G29 W I1 J4 Z-0.19250
Recv:13:57:20.005: echo: G29 W I2 J4 Z-0.17250
Recv:13:57:20.006: echo: G29 W I3 J4 Z-0.15500
Recv:13:57:20.008: echo: G29 W I4 J4 Z-0.19000
Recv:13:57:20.008: echo:Endstop adjustment:
Recv:13:57:20.009: echo: M666 Z0.00
Recv:13:57:20.011: echo:PID settings:
Recv:13:57:20.011: echo: M301 P17.39 I1.21 D62.34
Recv:13:57:20.013: echo:Z-Probe Offset (mm):
Recv:13:57:20.014: echo: M851 Z-3.40
Recv:13:57:20.014: echo:Linear Advance:
Recv:13:57:20.014: echo: M900 K0.00
Recv:13:57:20.016: echo:Filament load/unload lengths:
Recv:13:57:20.017: echo: M603 L538.00 U555.00
Recv:13:57:20.018: ok

First I had Heatbed PID heating enabled, when I see the issue. Then I switched to bang-bang heating with no difference. 
- Hotend temperature: 200 deg.
- Heatbed temperature: 60 deg. 
For the first minutes the temperature falls and rises by 1 deg, between 59 deg and 61 deg. Then the temperature falls gradually degree by degree til it reaches 56 deg. Then firmware was halted. The desired value was unchanged at 60 deg.

I did mods at my printer, added BLTouch and two MOSFET amplifiers for hotend and heatbed.
I tried to see what happens, when I heat up hotend to 200 deg and heatbed to 60 deg and have no print job running. Both temperatures keep stable over about half an hour. So I dont believe there is an overheating or a powerloss.
<p> Hope for some suggestions what to check.

davidramiro, 2021-01-24:
Not sure if this really is a firmware bug, especially when either PID or bang-bang heating render the same issue. 

Since you mention you modified the heatbed power circuitry, I'd rather take a look at that. I am also not familiar with the concept of a MOSFET amplifier? I use dedicated MOSFETs for the heatbed myself, to circumvent the rather small MOSFETs on the board, but I have never heard of an amplifier. The power from the PSU is more than strong enough to heat up the bed properly and keep the temperature.

As a last idea, and that might be far fetched, do you have upgraded your parts cooling fan? If it's a really strong one, it might actually cool down the bed enough, as the temperature sensor is right in the middle of the bed, where most of the prints would happen.

Gleis1234, 2021-01-25:
Thank you for your ideas. Amplifier may not be the right expression. It is a separate module with a with a big heatsink, to keep heat off from the mainboard.
[image]
I did this, because I could see the shutdown of the heater before I started upgrading my MEGA S. The first time I recognized the effect I used Marlin firmware 1.1.9 without activated BLTouch.
The parts cooling fan is the original one, which was shipped with the MEGA I3. Normally I reduced my fan speed to 75 %. May be I can try to position the print head in the middle of the bed an switch on the fan with 100 %.
Another Idea could be to exchange the two MOSFET modules for hotend an hotbed. What do you think about?

davidramiro, 2021-02-14:
I do have exactly the same MOSFET installed to bypass the heatbed MOSFETs on the board. No issues on my end - I am still very much in doubt about this being a software issue. Might even be something as simple as a faulty MOSFET unit you got. I'll tag this as a hardware issue and close this for now, but if you do find out anything that would fix this in software, make sure to report back.

closed 2021-02-14 by davidramiro