davidramiro/Marlin-Ai3M #111: [BUG] - Hotend cools down during printing, PID tuning doesn't help

wes1993, 2021-02-01:
Describe the bug
In preheat, hotend manages to heat up to 240°C and maintains stable. but when I start to print for a couple of minutes, it starts to cool down to 230°C and even lower. later the thermal protection seems to kick in and all motor freezes, the print just stops.
Notice : when it stops printing, the screen loses control, it cannot read the correct temperature. The pause/stop button can be clicked but it will just show please wait forever.

Tried M303 E0 S240 C15 U1 to tune hotend PID and nothing different happens.

To Reproduce
Steps to reproduce the behavior:

Start printer
Preheat ABS
Throw any gcode with above 220°C via SD card
A few minutes later, It just stops printing
Expected behavior
Expect it to print normally without cooling down.
I know it's very likely that I did someting wrong instead of a bug. Could you please help me out?

Printer (please complete the following information):

Firmware version: V1.4.6
Printing method: SD, sliced with Cura
Model: i3 mega with single Z endstop

Whith stock firmware everything work fine

M503 Output
Please paste the complete output of the M503 command here.

bassamanator, 2021-03-24:
I experienced this same issue twice yesterday though I was using PlA+ with temp set to 215C. I downgraded to the stock firmware this morning and I've been printing issue free all day long.

Firmware version: V1.4.6 (showed up as 116 in the About section of the printer)
Printing method: SD, sliced with Cura
Model: i3 mega S (QC card read August/September 2020)

wes1993, 2021-03-24:
Hello,
I have found the problem, it’s related to the speed of the fan, you should set the speed of the fan to max 70% in cura and you will print like a charm!! :)
Also an extra good step is to buy a sock for the hot end :)

bassamanator, 2021-03-24:
There is already a stock on my hotend and my fan was at 100% though :/

davidramiro, 2021-03-24:
This is mentioned in the FAQ. The stock hotend fan is quite powerful. Limit it to 70, like @wes1993 said, so it behaves just like on the stock firmware.

Just for reference, when printing PLA at 210°C, 50% on the fan is more than enough for me.

https://github.com/davidramiro/Marlin-AI3M/wiki/Frequently-Asked-Questions#my-print-suddenly-stopped

closed 2021-03-24 by davidramiro

bassamanator, 2021-03-24:
I've changed the fan speed to 70% in Cura. Doing a small print right now and it seems to be going just fine. 

Thanks @davidramiro! Your work on the firmware for this 3d printer is greatly appreciated.