davidramiro/Marlin-Ai3M #117: Unable to connect via USB after upgrade[BUG]

JonyBCN, 2021-04-18:
After installing the latest version (Marlin-Ai3M-v1.4.6-TMC2208.hex) everything has worked. The printer is working (after a scare because I left the cables connected as they come from the factory but you have to change those of the Z axis), but the impressions are correct.
  The problem has come when it comes to calibrating the extruder. Previously it connected to the printer via COM3. The port has disappeared and neither CURA nor Repetier are able to connect to the printer. ARDUINO ide also detects nothing. I have tried both Windows 10 and OSX with the same result.

smart734, 2021-04-18:
Hello, on Windows 10 check device manager, if there is COM device or unknown device. If unknown, try to reinstall CH341 driver. :-)

JonyBCN, 2021-04-18:
I just checked it on windows:
Unknown USB device (Port establishment error) 

Anyway I have installed the driver on windows and mac. Without result in any.