davidramiro/Marlin-Ai3M #126: STARTUP_CHIME Anycubic Meg

dietlow, 2021-12-21:
Hi I would like to change theSTARTUP_CHIME back to the original Anycubic STARTUP_CHIME. I have enabled and located the section in the file but for the life of me I can't figure out how to change it. 
Any help would be appreciated
Thanks

davidramiro, 2021-12-21:
Hi, do you mean restoring the original song after having enabled the `STARTUP_CHIME` flag?

The chime is defined in here:
https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/AnycubicTFT.cpp#L119

The original tune is to be found here:
https://github.com/ANYCUBIC-3D/I3-MEGA/blob/master/Marlin_TFT_CR_V1.1.1/music.cpp#L32

I haven't heard the tune in a long time, but I think the rhythm went something like this:

```
#ifdef STARTUP_CHIME
    buzzer.tone(150, 2794); // F7
    buzzer.tone(150, 2093); // C7
    buzzer.tone(150, 2217); // D7
    buzzer.tone(150, 2349); // E7
    buzzer.tone(300, 2217); // D7
    buzzer.tone(300, 3136); // G7
    buzzer.tone(600, 3520); // A7
#endif
```

dietlow, 2021-12-21:
Thank you very much I didn't expect you to respond so fast. I adjusted the tones slightly to better match the song you can see below but the problem I'm having I'm getting a pause at the third note any ideas what's causing the pause?
buzzer.tone(250, 2794); // F7
buzzer.tone(250, 2093); // C7
buzzer.tone(250, 2349); // D7
buzzer.tone(250, 2637); // E7
buzzer.tone(250, 2349); // D7
buzzer.tone(500, 3136); // G7
buzzer.tone(500, 3520); // A7
> On Dec 21, 2021, at 6:03 AM, David Ramiro ***@***.***> wrote:
> 
>   buzzer.tone(150, 2794); // F7
>     buzzer.tone(150, 2093); // C7
>     buzzer.tone(150, 2217); // D7
>     buzzer.tone(150, 2349); // E7
>     buzzer.tone(300, 2217); // D7
>     buzzer.tone(300, 3136); // G7
>     buzzer.tone(600, 3520); // A7