davidramiro/Marlin-Ai3M #131: [BUG] Lack of filament error with stock firmware 1.4.6 on Mega S

harry7922, 2022-07-27:
I installed the stock firmware for my Anycubic Mega S (no modifications).
Since the update I receive the error "lack of filament or filament monitor abnormal". The error only shown after 5-10 layers of perfect printing. Filament was not runout.

Don't know how to handle this issue. With original firmware, everything works fine and also filament sensor is fine.

Steps to reproduce the behavior:
1. Start printer
2. Print any model
3. Start printing
4. Wait for printing first layers
5. See error is shown in display and continue printing is not possible

This error should not occur since the filament really runout.

**Photos of the print / output of the terminal**
If applicable, add photos or console outputs to help explain your problem.

**Printer (please complete the following information):**
 - Firmware version: Marlin-Ai3M-v1.4.6-stock_drivers.hex
 - Printing method SD-Card
 - Model i3 Mega, Mega-S]

**M503 Output**
SENDING:M503
echo:  G21    ; (mm)
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E407.30
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z6.00 E60.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y2000 Z60 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R3000.00 T3000.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000 S0.00 T0.00 X10.00 Y10.00 Z0.40 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Mesh Bed Leveling:
echo:  M420 S0 Z0.00
echo:Endstop adjustment:
echo:  M666 Z0.00
echo:PID settings:
echo:  M301 P15.10 I0.93 D61.56
echo:  M304 P159.67 I29.99 D566.83
echo:Linear Advance:
echo:  M900 K0.00
echo:Filament load/unload lengths:
echo:  M603 L538.00 U555.00

Thanks for your help and great work!

Regardy, 
Hardy

Giova02, 2022-12-22:
Hi, have you found a solution?

harry7922, 2022-12-22:
No, I installed firmware of Knutwurst. Same issue there, but you can deactivate Filament Sensor in special menu.

Giova02, 2022-12-24:
Same here but I reinstalled the stock Anycubic I3 Mega S firmware thinking it was a firmware bug. Still getting the same filament sensor error so I unplugged the connector from the sensor and bypassed it with a small wire so the printer always read that there's filament in. Dont wanna spend money on a new sensor :)
Just have to pay attention to print with enough filament lol