davidramiro/Marlin-Ai3M #137: Extruder motor stops working[BUG]

PMarco03, 2023-03-26:
After 2 or 3 layers the extruder motor of my Anycubic I3 Mega S stops working with TMC 2208.
Once installed the firmware, i had to reverse the extruder motor's pin to make it working correctly. 
I tried to mash calibrating the printer different times, i checked the Vref of the drivers, i tried to swap the drivers, and i tried lowering the plate manually but the motor keeps loosing steps and than it stops working and i need to reboot the printer to make it work again.
I tried extruding filament with the built-in function and it work properly.
I followed the guide, i sent those commands after installing the firmware: 
- M502
- M500
- M92 E384
- M203 E30
- M204 R1500.00 
- M500 
I calibrated the plate manually, before using the "mash calibration" function but it still doesn't work.
I tried to reduce print speed at 40 mm/s but it still stops working.
Those are the driver i'm using: https://www.amazon.it/dp/B086MYZHD9?psc=1&ref=ppx_yo2ov_dt_b_product_details

davidramiro, 2023-03-26:
Print via USB and check the output when it stops. Might be runaway protection kicking in.

smart734, 2023-03-26:
Which slicer are you using? I had the same issue with new Prusa Slicer 2.5.0, so I tried Cura and it is working properly.

PMarco03, 2023-03-27:
> Print via USB and check the output when it stops. Might be runaway protection kicking in.

it doens't give any error messages. the printer just keep printing but without extruding the filament

> Which slicer are you using? I had the same issue with new Prusa Slicer 2.5.0, so I tried Cura and it is working properly.

Cura 4.11.0

Now I tried to update the firmware to marlin V2 but it has the same error. During a print i noticed that the motor became more noisy for 2 seconds and than it has stopped working again