davidramiro/Marlin-Ai3M #141: [BUG] "No worries. You can easily go back to the default firmware and restore the default settings." Not so much.

paulbeard, 2023-09-24:
**Describe the bug**
A clear and concise description of what the bug is.
The firmware didn't load or was perhaps incompatible with this unit (it has the blue/yellow display) as it is now bricked. Will power on but no functionallity, no heating, no stepper motors, connectivity through USB. Can it be re-flashed from the SD card? 

**To Reproduce**
Steps to reproduce the behavior:
1. Start printer
This is all it takes. The firmware didn't allow me to print, the SD card was not readable (some files showed up, bot not all and I couldn't make out if it was naming issue or what). I decided to drop back to the OG firmware and now the printer is bricked, no firmware version showing in the About panel. 

pronterface and cura cannot connect, they don't see a device at the end of the connection. The touchscreen works but few of the commands do. 

**Expected behavior**
A clear and concise description of what you expected to happen.

I expected it to print at least as well as it did with the OG firmware. 

**Photos of the print / output of the terminal**
If applicable, add photos or console outputs to help explain your problem.

I wish there was something to share here. 

**Printer (please complete the following information):**
 - Firmware version: [e.g. v19.02.08]
 - Printing method [e.g. SD, USB (OctoPrint, Cura, ...)]
 - Model [e.g. i3 Mega, Mega-S]

**M503 Output**
Please paste the complete output of the `M503` command here.

N/A. it can't communicate.

paulbeard, 2023-09-26:
> Is the red LED on the motherboard on?
>
> Try turning off the power supply, disconnecting USB, then setting the
> jumper on the motherboard from DC to USB and connecting USB again. This
> might get you a connection and the ability to flash.

The red light is on until/unless I move that jumper. No connection either
way.

The power supply fan blows regardless. Moving the jumper means no screen to
go with no red light. I have the little USB flashing dongle on its way to
me, in case that's all that's needed here. Is there anything else I can try
in the meantime?

paulbeard, 2023-10-02:
Any other thoughts here? I am looking at the menu LCD but nothing really
works. The only real win here is the end of that chime but right now I'd be
glad to hear it again. Neither Cura nor pronterface can reach it. I have
the little cable to access the chip directly and re-install a bootloader or
firmware but I don't know how optimistic I should feel about that. Is this
something other people have seen?

davidramiro, 2023-10-02:
Those Atmega boards are seldom bricked beyond repair, so a flash via serial has good chances of reviving it. That being said, just flashing a hex without bootloader should never have the board lose the ability to flash via USB like that, so either something went wrong while flashing or there was some preexisting defect.

This is actually the first report of a "brick" like that if you search through the issues on here, so I'm really not sure how to help. Are you positive that you used the correct original firmware? There are many versions of the i3 Mega out there nowadays and the 2 versions I provide on my webspace might be outdated.

davidramiro, 2023-10-02:
https://www.anycubic.com/pages/firmware-software

Worth noting that there are driver packages provided for each version of the printer, so maybe that might be worth looking into.

paulbeard, 2023-10-02:
Thanks for the reply. I pulled the firmware from AnyCubic, I think, but the
real problems were with whatever I updated with. I think
I picked the right hex but maybe I missed something. There are several
variants of this model, as it turns out. The firmware update was when the
touchscreen became inop and where I can't reconnect via USB. Wishing there
was something like the hard reset sequence on open
firmware compliant routers as an option here…been down that road many
times. When that serial cable arrives, I'll see if I get some resolution.
My best bet may be to get a modern motherboard for it, and skip all this.
Best case is if I have a stock anycubic mega s and after running it
alongside a Prusa, I have to think about that.

I'll look at the anycubic site again to see if I missed anything or if
there is any update.

paulbeard, 2023-10-04:
Well, I seem to be at an impasse here. I can connect the board via a
USBtiny dongle to a machine that supports avrdude. I get a green light on
the USBtiny device and power LED on the trigorilla board but finding the
serial port in macOS is beyond my skills. Lots of searches and nothing had
worked: the USBtiny shows  I assume the USBtiny is connected properly or
the LEDs wouldn't light up. I'll have to look around to find out how to get
this all working. This obviously precludes buying an upgrade board since I
would be in the same situation with it.

I realized this is beyond the scope of the firmware development process but
maybe you can think of some way to unbrick this thing or get another
working motherboard. Maybe the board itself is damaged in some way, I have
no way to tell. It does boot and display the UI but nothing except changing
pages within it.

ioreg -p IOUSB -l -b | grep -E "@|PortNum|USB Serial Number" will
reveal the USBtiny…
    +-o ***@***.***  <class IOUSBHostDevice, id 0x1000a62fc,
registered, matched, active, busy 0 (5 ms), retain 18>
but nothing useful appears in /dev

Thanks for any info you can share.

paulbeard, 2023-10-05:
Well, after a lot of faffing about, I was able to get the stock firmware
reinstalled. Pretty sure I have followed the exact same steps at least once
before but we've all been there.

closed 2023-10-08 by davidramiro

paulbeard, 2023-10-08:
There were a few lingering problems to sort out…the heating/thermistor wires were disconnected (they were glued to the display daughterboard so when I pulled that to get to the serial interface, they came with it. Figuring out where they went took some doing.) Anyway, it seems to be 100% operational now, wrapping up an 11 hour print job. 

So I am back where I was two weeks ago. But a little more cautious and somewhat mystified. There are quite a few firmware options here, some very specific to the revision of this machine. and the ability to connect via serial to send the reset commands is not very reliable but did work yesterday. Advice? Guidance? Should I even persist in this, given the varied models of this machine?